hax0kartik/rehid #1: Is mikewii's latest 3gx loader supported?

Mewthree3, 2021-03-27:
(Sorry for not using GBAtemp, I can't use it, so I'll just ask here.)
Is [this](https://github.com/mikewii/Luma3DS-plugin/releases/tag/r1.0.10) supported?

Mewthree3, 2021-03-27:
I just realized that it might be a plg loader. Sorry if it is

Mewthree3, 2021-03-27:
Even if that wasn't one, I also tried it with Nanquitas' loader, and it doesn't work with it as well.
(https://github.com/Nanquitas/Luma3DS/releases/tag/v10.2.1)

hax0kartik, 2021-03-27:
Use the release from the second page of gbatemp and check.

Mewthree3, 2021-03-27:
It works fine now, thank you.
However, I'm also facing the same issue as @NanashiFinal (on GBAtemp), where I can't use remap buttons on games that don't support them (I've tried out some hb, new super mario bros 2 (gold), and an indie game called "toy stunt bike").
I can't speak for the other buttons working or not, because I onlu use c-stick remapping and zl/zr remapping.

hax0kartik, 2021-03-27:
@Mewthree3 there are some test builds in my discord server which you could try. Please tell me if they work.  https://discord.gg/aGcfmzW6

Mewthree3, 2021-03-27:
I can't access discord (this is a huge pain for you, apologies about that)
Could you upload them here, or add them to a gdrive or just email me? (
***@***.***)
Thanks again, apologies for the nuisance

hax0kartik, 2021-03-27:
Please try this. For some games, it has an input lag for others it just works fine. Please check if your work fine.
Also the json file needs to be updated, please check the second page of gbatemp thread to update it.
[0004013000001D02.zip](https://github.com/hax0kartik/rehid/files/6215024/0004013000001D02.zip)

hax0kartik, 2021-03-29:
Closing due to inactivity and original issue being fixed.

closed 2021-03-29 by hax0kartik