hax0kartik/rehid #4: Bug: Touchscreen positions scaled down?

SchusterEric, 2021-05-20:
System information:
Old 2DS
Luma 3ds cfw (latest version)
System version latest

Problem:
When I have rehid installed, all touchscreen inputs will be incorrect, except at the centre of the screen. As the actual position I touch moves towards an edge/side the registered input will first seem to match, but get increasingly more inaccurate the more it is away from the centre of the screen. When the actual (pressed) position reaches a side, the console will register the input to be ca. 1mm away towards the centre of the screen.

Here is a screenshot that shows this in effect (made in the build in notes application), I drew this by pressing against the frame of the screen with a stylus. The red line was drawn without rehid installed (The white area around it is actually not visible on the console itself, it might have been added in the export) and it perfectly matches up with the actual frame of the screen on my console. The blue line was drawn with rehid installed (And no remapping profiles active) and the resulting rectangle (minus the bottom) is way smaller than it should be. 

[image]

With rehid installed, I can not draw in the area between the blue and red line, which I could when it was not installed.
The because the (registered) inputs tend towards the centre on all 4 sides and get linearly more inaccurate it seems to me, that maybe rehid assumes that my screen/touchscreen is smaller/larger than it actually is.

hax0kartik, 2021-05-21:
[0004013000001D02.zip](https://github.com/hax0kartik/rehid/files/6520880/0004013000001D02.zip)
Please check this build and see if the problem is fixed.

SchusterEric, 2021-05-21:
Yes, that build fixes the issue

hax0kartik, 2021-05-21:
Great! Thank you for reporting.

closed 2021-05-21 by hax0kartik