hax0kartik/rehid #7: Bug

itchyd, 2021-06-11:
When I close the lid for awhile leaving the console on and come back to my game (this is dqm2) it sometimes gets in this state.  No buttons do anything except the home button but that is no help because even on the home screen I can't do anything.

It's a big problem because I can't save my game when it's in this state.

itchyd, 2021-06-11:
The power button must be pressed 2x to turn the console off and escape this state.

hax0kartik, 2021-06-11:
[0004013000001D02.zip](https://github.com/hax0kartik/rehid/files/6638171/0004013000001D02.zip)
Try this out, extract the contents of  the zip to `/luma/titles/`

itchyd, 2021-06-17:
Sorry it's hard to prove a negative, and I just installed this.  Let me try it for a few days.

itchyd, 2021-06-17:
Nope that did not fix it.  All I have to do is open and close the system a few times.  Here's the remaps I'm using.  The game is Dragon Quest Monsters 2 fan translation.  

[image]

hax0kartik, 2022-06-11:
Seemingly fixed in latest commit.

closed 2022-06-11 by hax0kartik