hax0kartik/rehid #13: C-nub to nothing?

AnityEx, 2022-01-28:
Hey I know this might not be the best place to ask something but, I've got a problematic c-stick that goes in random directions for no reason, so is there a way to remap something to not be registered at all? Would be so happy if this were the case I cannot play binding of Isaac anymore bc of this it's stuck to down sometimes and other times up

hax0kartik, 2022-01-28:
You'll need a ir patch, join the discord server and I'll make one.

AnityEx, 2022-01-28:
Oh, thanks! may i have the invitation link? i dont seem to find it

hax0kartik, 2022-01-28:
Yup, https://discord.gg/hyuvmb9

AnityEx, 2022-01-28:
i got to wait 10 minutes in your seerver to talk, wait a second

AnityEx, 2022-01-28:
**Binding of isaac has been restored** thank you for the help it means a lot im serious, this is my only 3ds and i cannot live without it agradecido con el de arriba as we say in my country, thanks!

closed 2022-01-28 by AnityEx