hax0kartik/rehid #18: Ability to apply rehid remaps to every title id

Gudak0, 2022-08-20:
If its built in how do I do this, else, can there be an option to apply a generic rehid remap to all titles and then custom ones for specific titles override the global remap? Would like this to be implemented as a workaround to me having broken L and ZL buttons

0xb01u, 2022-11-12:
This is already built in. You activate a global remap by placing `rehid.json` inside the folder `rehid` directly, not in any subfolder.
When using rehidhelper, after scanning a QR code, the very last title on the title list to apply the remap to should look like a blue "G". That stands for "global", so choosing that "title" will apply the remap to the whole system, unless a title-specific remap exists for the current title.

However, I just noticed rehid doesn't seem to work anymore. Probably because of some recent update. I will open an issue about it, luckily I will get a response by the developer.

0xb01u, 2023-01-03:
Hi again.

I just noticed that this repository was updated a few weeks back, fixing the issue I linked in my previous comment. Rehid should be working properly again, including global remappings, as described in my previous comment.

Please note that there is no release for that updated version of rehid yet, so you would have to build and install it manually.

lisiyaki, 2023-07-28:
This is still not working... is the newest release not the updated version talked about before?

closed 2023-07-28 by hax0kartik