hax0kartik/rehid #19: Rehid stopped working completely

0xb01u, 2022-11-12:
A few months back I updated my 3DS to the latest system version, 11.16.0-49. I checked beforehand that cwf/luma did not break, and after updating everything seemed to work fine.

However, just today I booted a game that used rehid, and I noticed that rehid is not working at all: the default 3DS mappings are always used. I tried creating new mappings, both for the game and for the global system, but no luck,

I bring up the new system update because that's the only thing I recall doing since rehid last worked on my device that could break it. It seems that this update is from 12th September 2022.

0xb01u, 2023-01-03:
Commit 45e3448ce007c616415027eda5f057e02ee25069 fixes this issue.

However, as it is right now, there is no **release** for a version of rehid including that commit. Thus, **to fix rehid**, the user has to download and compile the source code (including getting its dependencies: devkitarm, ctrulib, ctrtool), install the `0004013000001D02` directory manually into the 3DS (inside `/luma/titles/` in the SD card), and also enabling game patching via luma (either from Luma3DS configuration menu, or by manually editing `config.ini` inside `/luma`). This process is quite a hassle.

I would eagerly **suggest to make a new release** with the updated rehid and rehidhelper, so that updating rehid and using the remappings again is more manageable for everybody.

closed 2023-01-03 by 0xb01u

0xb01u, 2023-02-15:
> 

Because I didn't know about that and that is not explained anywhere in the repository either.

With this new piece of information, the repository should do one of the following:
1. Make a new release with the updated version of the program, so that the latest release in the Releases page works with the latest version of Luma.
2. Clearly indicate that builds are automatically created for every commit and where to find them.

hax0kartik, 2023-02-16:
I'll soon update the repo and mark a new release