hax0kartik/rehid #21: rehid not working Luma 12.0

tak914, 2023-03-16:
when upgrading to luma 12.0 rehid stopped working

0xb01u, 2023-04-12:
I can confirm that toggling on "Enable loading external FIRMs and modules" in the Luma3ds menu makes rehid work again.

By the way, I can see that the current (latest commit) code of rehid automatically toggles on that option. I guess this is an issue that arises from using older versions of rehid.

Ward777, 2023-05-22:
I have the same problem when upgrading to Luma 12.0, I tried to turn on the "Enable loading external FIRMs and modules" option, rehid still not working..

Cimitri, 2023-07-02:
Same here. Rehid still not working even after enabling the option in Luma. Please help ! I have a broken R button in a n2ds xl and really wanted it to remap it to ZR.

KekkoGH, 2023-07-08:
> Same here. Rehid still not working even after enabling the option in Luma. Please help ! I have a broken R button in a n2ds xl and really wanted it to remap it to ZR.

I'm trying to get it to work since yesterday but it's the same for me too... no sign of success.

EmeraldTuxedo56, 2023-07-23:
Still Not Working On New 3dsXL I Just Downloaded The Latest Update And Also Luma's Latest Update And Also Enabled External Firms Too And It Still Doesn't Work Even After Enabling It In The App

hax0kartik, 2023-07-23:
> Still Not Working On New 3dsXL I Just Downloaded The Latest Update And Also Luma's Latest Update And Also Enabled External Firms Too And It Still Doesn't Work Even After Enabling It In The App

Join my discord and ping me, I'll need some more context to help you out