hax0kartik/rehid #25: Rehid Error When installing causing 3DS buttons to not work on home screen

nshannon123, 2023-08-05:
Same issue noted by BuTsu in the GBA Temp thread: https://gbatemp.net/threads/wip-rehid-button-remapping-for-3ds.585387/post-10118204

Updated from old ReHid via Homebrew menu to use the CIA method. Installed RehidHelper onto Home Screen - went into it and tried to install Rehid from the app on the 3DS. This caused 3DS to crash and now upon restarting no buttons respond.

Can rectify the problem slightly by disabling loading external FIRMs and modules on Luma settings but when trying to open any game or application will not work. A button will not work and using touch screen to open any application will cause the 3DS to freeze on home screen. Any idea what's happened and if there is a fix?

hax0kartik, 2023-08-05:
Please send me a photo of your luma config screen.

nshannon123, 2023-08-05:
No problem
[image]

hax0kartik, 2023-08-05:
Delete the /luma/titles/0004013000001D02 and /luma/titles/0004013000003302 folder manually

nshannon123, 2023-08-05:
I deleted both folders before installing the CIA as per the release instructions

hax0kartik, 2023-08-05:
Hmm, make a copy of rehid folder to your desktop and then delete that. And delete the `0004013000001D02` `0004013000003302`  in luma/sysmodules as well.

hax0kartik, 2023-08-05:
After that, make sure you have the latest version of rehidhelper and re install rehid and then try generating and using your remappings again.

nshannon123, 2023-08-05:
Ok - some progress after deleting the files from sysmodules - can use buttons again with loading external FIRMs and modules checked at least. However A button not working for opening anything on the home menu. Using touch screen to open an app causes the 3DS to be froze on home screen still.

It feels as if something has gotten corrupted or something when the system crashed initially when installing rehid from the application. Even the wifi indicator, and time etc are missing from the top screen on the home menu. I have failed to do anything on the system except access luma or GodMode9 and navigate the home menu.

hax0kartik, 2023-08-05:
It is impossible to corrupt your system usijg rehid/rehidhelper.
Join the discord https://discord.gg/hyuvmb9 so that we can diagnose this issue further ^^

hax0kartik, 2023-08-06:
Issue resolved by user on their own.

closed 2023-08-06 by hax0kartik