hax0kartik/rehid #27: After closing 3ds in sleep mode, remaps don

NerdyCafe, 2023-09-01:
I was playing a game with the remapping and had closed my ds. Whenever I reopened it all the buttons stopped working. After I closed and reopened it the buttons started working again, all except for the remapping I had. (ZL ->L and ZR ->R) I've tried to remap them but it doesn't seem to work. Everything had worked before, I'm just not sure why the remapping isn't working now.

hax0kartik, 2023-12-31:
Hi, apologies for the late reply. I fixed a similar-ish bug a little while ago. Would you mind joining my discord server so I can help debug this issue?

NerdyCafe, 2023-12-31:
You're all good. I'm already in the discord under, I'll text there and we can go over it.

For anyone reading the thread, what I ended up doing was deleting my original remaps and then remapping them. It's worked since then, although I can't give any way to avoid this bug since it seems random.