hax0kartik/rehid #28: Attempting to toggle rehid always results in "Disabled" message

Mine4114, 2023-10-09:
I accidentally toggled rehid, and now it tells me it's disabled even when toggling on. Process list confirms that rehid is not running.

TGRush, 2024-05-14:
Bump! I've found no possible reason as to why this is happening, wonder if more people are experiencing the issue

hax0kartik, 2024-05-14:
1. Check if luma is updated.
2. Check whether rehid is forrectly installed.
3. Check if rehidhelper is the latest version.

EternityEngine, 2024-07-20:
I am also experiencing this issue.

1. Currently updated to Luma3DS v13.1.2.
2. Downloaded latest release of rehid through RehidHelper.
3. Reinstalled v4.0 from the QR code on the release page.

I'd try a complete uninstall and reinstall but I'm not sure of the exact way to go about that.

Mine4114, 2024-07-20:
I seem to have found the cause, it seems that opening the helper automatically enables rehid for me.

EternityEngine, 2024-07-20:
The toggle doesn't actually seem to be disabling rehid either, since my remapping for Kid Icarus: Uprising only stops working if I rename the folder for it in `rehid`.