hax0kartik/rehid #32: RehidHelper generates wrong title IDs

ItsCrocoSwine, 2024-03-27:
RehidHelper v4.0 generates the wrong title IDs when scanning a QR code and selecting a title. It doesn't seem to be all of them, just most of the titles later in the list generate the ID of the previous title in the list instead.

MidyGamy, 2024-05-02:
I can confirm it

ChrisWinters-CA, 2024-05-15:
Also confirming.

Ocarina of Time 3D should have the following TitleID:
0004000000033500

However RehidHelper generates the following TitleID instead:
000400000BEEF500

For now it can be worked around by simply renaming the folder with the correct TitleID.

hax0kartik, 2024-05-16:
> Also confirming.
> 
> Ocarina of Time 3D should have the following TitleID: 0004000000033500
> 
> However RehidHelper generates the following TitleID instead: 000400000BEEF500
> 
> For now it can be worked around by simply renaming the folder with the correct TitleID.

Hi can you please upload a screenshot of the game selection screen in rehidhelper? Thanks!

ChrisWinters-CA, 2024-05-16:
Sure. I completely uninstalled and reinstalled Rehid through RehidHelper to make sure I'm starting with a completely default configuration, just as anyone would when installing for the first time. I also included a screenshot of 3DShell showing the incorrect TitleID after scanning the QR code and applying it through RehidHelper.

There is another page of games and applications, I just highlighted OoT3D. Let me know if you need a screenshot of the first page as well.

[image]
[image]

ChrisWinters-CA, 2024-05-16:
Also adding that I just checked through the title list in FBI and it appears 000400000BEEF500 is the TitleID for ftpd pro. Not sure if that's helpful.

hax0kartik, 2024-05-17:
> Sure. I completely uninstalled and reinstalled Rehid to make sure I'm starting with a completely default configuration, just as anyone would when installing for the first time. I also included a screenshot of 3DShell showing the incorrect TitleID after scanning the QR code and applying it through RehidHelper.
> 
> There is another page of games and applications, I just highlighted OoT3D. Let me know if you need a screenshot of the first page as well.
> 
> ![3DShell](https://private-user-images.githubusercontent.com/15055872/331274307-c441dfcc-c41c-4ef6-9b58-207cc75a7a07.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU5NTc2MDAsIm5iZiI6MTcxNTk1NzMwMCwicGF0aCI6Ii8xNTA1NTg3Mi8zMzEyNzQzMDctYzQ0MWRmY2MtYzQxYy00ZWY2LTliNTgtMjA3Y2M3NWE3YTA3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE3VDE0NDgyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA5YmZjNDgzODhjZTE0MzRhZGQ3MGZmNTdjMmY3MjI5NmU5Mzc1MWRkNTM3ZDZkOWNlOWQ0YTc4OTYxMWEwZTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.YZgMqokb_CTSNqOdVjjuHQN0RBi5ZoabZtrs1_Nz0_Y) ![RehidHelper](https://private-user-images.githubusercontent.com/15055872/331274310-4cb9ba8e-5d3f-4eec-9d41-076e976a3489.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU5NTc2MDAsIm5iZiI6MTcxNTk1NzMwMCwicGF0aCI6Ii8xNTA1NTg3Mi8zMzEyNzQzMTAtNGNiOWJhOGUtNWQzZi00ZWVjLTlkNDEtMDc2ZTk3NmEzNDg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE3VDE0NDgyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZjNWE5MTVhZTM5MmEzNmU0NDc0Y2EyMzZjNDNkZmViNWQ2MmI4MWI4MDkzMDcwNDk5YzI4NjUwMjJiMDgwMTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AwL4z6JGoKsARrcNY1dDunBTR4lYEwVbyi-LgJvf_Mg)

Hi, can you send me the first page as well? It is pretty clear that there is a off by one error somewhere in my code.

ChrisWinters-CA, 2024-05-17:
Joined the discord and here are both pages

Page 1
[image]

Page 2
[image]

ChrisWinters-CA, 2024-05-20:
After some trial and error bug hunting we think the issue might be related to some apps that aren't getting their icons generated correctly in RehidHelper. In my case I have 3 emulator apps that do NOT show up in the game list in RehidHelper after scanning a QR code.

temperPCE
virtuaNES
picodrive

When selecting an app to generate the remap file it is grabbing the TitleID of the 3rd game/app to the left of the selected one. Precisely the number of apps installed on my N3DSXL, but not being displayed.

The TitleIDs for these emulators are in the list, but the icons aren't being generated correctly. Kartik is working on a fix. For now, just rename the folders to the correct TitleID as a workaround.

hax0kartik, 2024-05-20:
[rehidhelper.zip](https://github.com/hax0kartik/rehid/files/15379857/rehidhelper.zip) Hi, please unzip this file and install the cia within it. Let me know if the issue is fixed.

MidyGamy, 2024-05-24:
> [rehidhelper.zip](https://github.com/hax0kartik/rehid/files/15379857/rehidhelper.zip) Hi, please unzip this file and install the cia within it. Let me know if the issue is fixed.

Seems to work for me

hax0kartik, 2024-05-27:
Closing as both @MidyGamy and @ChrisWinters-CA have confirmed the fix to work.

closed 2024-05-27 by hax0kartik