hax0kartik/rehid #33: D-Pad to C-Pad broken

Trickygamers, 2024-04-06:
when I enable D-Pad to C-Pad the D-Pad will do it's default with the C-Pad on top of  it. esentially the D-Pad and C-Pad are  being pressed at the same time like when I press up I am mved twice instead of twice.

hax0kartik, 2024-05-20:
That is a delibrate design choice. There are some people who want the exact behaviour you are describing. IIRC, some people use rehid with mk7 where they want to use the dpad as well as cpad to control the kart.

I wonder if there are some sort of rectifcation i can do for this.. maybe introduce a new entry such as `dpadtocpadnodpad:True`? I am open to other ideas as well if you want to suggest any ^^

Trickiy, 2024-05-20:
What I would do is make a new entry like you are already doing but rename it to D-pad=C-pad or something else that is dummy proof. Because some people like myself are idiots.

ipkpjersi, 2024-10-07:
Hi  @hax0kartik I think I might be experiencing this issue

When I play Smash Bros 3DS with `{"dpadtocpad":true}` when I move the dpad, it performs a taunt rather than moving my character - my character only moves if I'm in the air and then press on a dpad direction.

I'd love to have a new option ` {"dpadtocpadnodpad":true}` and honestly I'd even be willing to donate some money on your ko-fi if that helped prioritize this feature, since to me, the idea of being able to override the control stick without having it duplicate inputs would be perfect and I know how thankless open-source work can be.

I'd love to hear your thoughts on this.

ipkpjersi, 2024-10-20:
Apologies for the double post but I'm curious to see if the workaround in my post works for you @Trickiy someone told me about it on GBAtemp.

Trickiy, 2024-10-22:
> Apologies for the double post but I'm curious to see if the workaround in my post works for you @Trickiy someone told me about it on GBAtemp.

I am sorry for the late response but my 3ds is broken at the moment so I currently can't test.

Funnykittens873, 2025-06-19:
> {"cpad":[{"press":"LEFT","get":[-190,0]},{"press":"RIGHT","get":[190,0]},{"press":"UP","get":[0,190]},{"press":"DOWN","get":[0,-190]},{"press":"DOWN+LEFT","get":[-190,-190]},{"press":"DOWN+RIGHT","get":[190,-190]},{"press":"UP+RIGHT","get":[190,190]},{"press":"UP+LEFT","get":[-190,190]}]}
> I tested this out, and it works exactly the way I wanted it to, I can play Smash Bros 3DS without the dpad performing taunt moves.

I've noticed a problem. Most games like Animal Crossing: New Leaf and Super Mario 3D world have a problem where you can play the game just fine but when trying to make selections in the title or dialogs, it won't even work at all.