hax0kartik/rehid #38: Circle Pad Pro (o3DS) Support (Feature Request)

Mayonaka-7, 2024-05-31:
It would be great if there were support for the Circle Pad Pro on o3DS and o3DS XL models.

N0RDICST0RM, 2024-09-09:
Commenting to add support. That would be sick.

sasiccia84, 2025-04-09:
Hi all! Are there any news on the circlepad support? Thanks a lot!

N0RDICST0RM, 2025-06-04:
Last time I checked, it isn't supported. Apparently, the Circle Pad Pro doesn't give any input at all except in games that already support it.

Mayonaka-7, 2025-09-03:
It looks appears that this was added to Red Viper in 1.0. I'm curious if a similar approach would work with rehid. 

[Red Viper commit](https://github.com/skyfloogle/red-viper/issues/68)

TheBerZerK, 2026-02-06:
the discord link is expired can i get a new one i would love to test it out