hax0kartik/rehid #39: [Feature request] Changes in turbo command activation

joaoviss, 2024-07-05:
First of all, thank you very much for the app, I have been looking for something like this for a long time. You see, I have motor coordination problems and can't get past parts of games that require quickly pressing a button. These games would be fun for me with a turbo controller.

I needed to use the turbo function for just one button in a game, so I created the file /rehid/<titleId>/rehid.json like this:
```json
{
	"turbo": [
		{
			"press": "ZR",
			"get": "X",
			"framedelay": 3
		}
	]
}
```
It works, but I have some observations. When I press ZR, it starts to press X repeatedly, even if I don't press anything else until I press ZR again. It works as if it were an on/off switch; I think it would be more useful if it worked as a "hold" function. You hold the turbo hotkey and press the button you want to rapid fire; the buttons work normally if you don't hold the hotkey. Also, this way, you would only need one hotkey in the json file and it would work on any other button without occupying any other button. I'm not sure how feasible any of this is, but perhaps the on/off switch issue could be possible.

hax0kartik, 2024-07-06:
Try with a framedelay of 0

xy172, 2025-09-05:
Originally I had this same issue but with frame delay of 1, tried frame delay of 0 and it just presses the button 1 time upon release of the "press" key.