hax0kartik/rehid #41: Rehid not selecting title (nor fuctioning)

fugi-8, 2024-08-26:
Pretty self explanatory, when I scan the qr code and go to the title select screen it just doesnt, I also tried the manual method by puting in rehid a folder with the game id and the rehidbuild file inside and didnt work, tried for rayman 3d to be more especific. Also tried reinstaling by deleting the files and etc but didnt solve the issue at all.
Sorry for the half baked english, it's not my first lango

bwarf79, 2024-10-07:
under luma 13.2.1 it seems to be broken, I reverted to luma 13.0.2 and it works again (I put the json in root\rehid to work with all games and menu)

elfeon-g, 2024-10-12:
> under luma 13.2.1 it seems to be broken, I reverted to luma 13.0.2 and it works again (I put the json in root\rehid to work with all games and menu)

Did you make sure that `Enable loading external FIRMs and modules` was enabled in your Luma settings?

I just had my rehid break after updating to Luma 13.2.1 too — and since it had been a while since I last updated Luma at all, I panicked a little and found your comment here, which led me to try downgrading as well to fix it... except that after a bunch of fiddling, I reread [this wiki page](https://wiki.hacks.guide/wiki/3DS:Rehid#Installation) more carefully, and found out the culprit was actually that setting needing to be enabled — even though it is disabled by default.
I have now upgraded back to 13.2.1, with the setting and rehid enabled, and it works with no issues ^_^

EmblemMicaiah, 2024-12-28:
> Did you make sure that `Enable loading external FIRMs and modules` was enabled in your Luma settings?
>
> I just had my rehid break after updating to Luma 13.2.1 too — and since it had been a while [...]

Sorry if this is kinda a dumb question, but how do you access the settings for luma on the 3ds? I haven't really messed around with anything like this since I initially homebrewed my 3ds back in 2020.

elfeon-g, 2025-01-12:
> Sorry if this is kinda a dumb question, but how do you access the settings for luma on the 3ds? I haven't really messed around with anything like this since I initially homebrewed my 3ds back in 2020.

It's mentioned on the page I linked in my previous comment (11. in Instructions), but you can access the Luma settings by powering off the console, and then holding `select` while you turn it back on! (if you hold `start` it will launch [GodMode9](https://wiki.hacks.guide/wiki/3DS:GodMode9/Usage) instead)