hax0kartik/rehid #43: ZR triggering original "ZR" press as well as remap.

Dizzy611, 2024-11-01:
I set up a global remap from ZR to B (and B to ZR) because very few games use ZL or ZR and my B button takes a lot of pressure before it responds. However, after using 3DS-TEST I noticed that, while pressing (mashing, really) B does simply show up as ZR, pressing ZR shows up as both ZR and B, which is not what I wanted. This won't be a big problem in most games which ignore ZR/ZL, but it is a bit annoying on the home menu. I see that this project hasn't really had any commits in six months or so, any chance of a workaround or fix?

Dizzy611, 2024-11-01:
This is my rehid.json for reference:

[rehid.json](https://github.com/user-attachments/files/17599454/rehid.json)

Dizzy611, 2024-11-01:
I just had the thought... is it executing the remappings in order? That might explain it, since "press ZR get B" is before "press B get ZR"?

hax0kartik, 2025-01-24:
Message me on discord, this has been a long time issue and I am certainly interested in fixing it. But i'll need some help with running the tests.

closed 2025-01-25 by hax0kartik