hax0kartik/rehid #48: Not applying mapping

Robbielewis, 2025-01-06:
After scanning my QR code and getting to the application selection screen, I am unable to select my game to remap, the A button does nothing. I tried a solution I found on reddit where you turn the Luma settings for loading external firms and game patching on and off again, but that did not resolve the issue for me.

Darthagnon, 2025-02-16:
I've noticed that rehid mappings keep getting disabled for me, too. I use it to map SELECT to the touchscreen for Monster Hunter 4U, and it works once or twice, then disables for the rest of the play session until I restart the game

Darthagnon, 2025-02-21:
Turns out I still had reHID in `SD:/luma/titles/0004013000001D02/blah.cxi`, whereas the newsest versions need to be placed in `SD:/luma/sysmodules/0004013000001D02.cxi`

tr1p0p, 2025-03-29:
Yeah same here. Non reaction when pressing A...
Why ?