hax0kartik/rehid #51: Rehid causing issue with sleep and the C-stick

Icing-weaponry, 2025-02-28:
3ds model: New 3ds (JPN converted to NSTC)
Luma version: 13.3
firmware: 11.17.0-50U
rehid installed: latest via rehidhelper (v4 I think)

Issue:
If the system is put to sleep and woken up while in game, it will lose its connection to the c-stick. The issue isn't every time though it seems to happen randomly and usually take 5-6 sleep before it manifest but then the only fix is a reboot of the system.
It only happens while in game, if the game loses the connection the home screen can still use/see it but other games will not see it either.

Testing:
I updated rehid to the latest version and I was able to reproduce the issue with putting my system to sleep 4 times.
I fully uninstall rehid removing the files in /luma/sysmodules 0004013000001D02.cxi, 0004013000003302.ips and rehid.cxi.
I also uninstalled rehidhelper and rebooted. Afterward I can put my system to sleep many many many times but the game never lose the connection to the c-stick. Not a great testing methodology but enough to lead me to believe that rehid might be the cause.

Do let me know if you require more information

thanks!

hax0kartik, 2025-12-14:
Hi, thanks for all the details! 

Could you join this discord server so we can further discuss this issue? 

https://discord.gg/hyuvmb9

Thanks!

hax0kartik, 2026-01-16:
This was confirmed to be not present on matser by @Icing-weaponry. Closing.